iscumd/UM-Dearborn-Mars-Rover-Challenge #12: Side Panel Research

kylesaenz, 2016-03-10:
Find a way to create lighter panels that can still be used to mount electronics onto them. look into aluminum, thin plastic, a combination of the two or more, etc.

CrisAdam, 2016-03-26:
@ (Jason) - why did we go with the seaboard over aluminum side panels originally?

jsspurlo, 2016-04-16:
We went with seaboard originally because it worked well for us in the past on other robots. It doesn't bend easy or have permanent dents that are noticeable unless you purposely hit it. It's good to spray paint, compared to aluminum, where we'd have to powder coat it for it to match the frame. We got 1/4" thick because we didn't want to to bend too much with the weight of the electronics being held on them.
 
If you did the weight calculations and determined that aluminum panels would be lighter and/or more effective, than I would be happy to have aluminum panels.

CrisAdam, 2016-04-16:
frame isn't grounded, and as far as I know all of the electronics being mounted on it have a large heatsink that is attached to the frame, i.e. there aren't exposed parts that can be shorted

also, "because it worked well for us in the past" is not a valid design reason for anything, lets not use it going forward

jsspurlo, 2016-04-16:
Why wouldn't it be a valid reason? Experience provides valuable advice. If it isn't broken, don't fit it. 
What I am saying is, if it is a better alternative to use, because you've used it before and it works for what we need it to, because you know there will be no problems, then it is a **definite good option**. If we don't have experience with it, but theoretically it is an improvement, then it is an **indefinite good option**. In this case, I evaluated that we would most likely have less problems going with seaboard than aluminum panels. If I knew aluminum panels would be a better option, then I would go with that, but I don't know.
 
What are the pros and cons of getting aluminum panels instead of seaboard? If it saves us weight, how much weight does it save? Is it worth the money to buy the aluminum? How easy is it to get it to the right shape we want? Who will make the panels? Are they skilled enough? How will the panels bend once electronics are mounted to them? Will that bending compromise function? 
And that's only the known unknowns.

CrisAdam, 2016-04-17:
@jsspurlo The first part of the design process is to come up with a design that meets the requirements - in order to successfully accomplish this it is essential to not limit ourselves by looking at previous works, and instead focus on the actual requirements (weight-savings, etc.) - then once we have designs that meet the requirements, we can do a design review process where we select the best design, and only then do we worry about actual implementation issues (i.e. cost, experience, etc.) - otherwise we will quickly end up proficient at creating obsolete robots.

jsspurlo, 2016-04-17:
@CrisAdam Yes, I am not disputing that. This was before we had any official design process. Nonetheless, I did follow the design process, I considered using other materials for the panels, and with the information I had at the time, decided that seaboard was the best option. If new information were to be made available that would make aluminum panels more appealing, then I would go with those no problem. Looking at previous works does not limit our capability to successfully design a robot that can accomplish the tasks, it does gives us valuable insight, because it teaches us ahead of time about what works and what doesn't. 
Now, I've been designing and building robots for the last 7 years, and designed the first robot in ISC to get first place in an international competition. I know the best options when I see them, because I have effectively trained myself over the years to take everything into account. As the Design Captain of Team Argon, I would hope that you respect my position to have made these decisions.

jsspurlo, 2016-04-18:
@CrisAdam Sorry, I didn't see the edits (2nd half of your above post) until now. Sure, now that we have this information, switching to aluminum panels looks like a theoretically good idea. But, we should only switch out the panels only if we find out we are overweight. Right now we need to focus on getting the wireless E-stop working, doing a test drive, implementing the signal light, making a pause switch, etc. At this point in time, the panels on the side of the robot is the least of my concerns.

closed 2016-05-22 by CrisAdam