iscumd/UM-Dearborn-Mars-Rover-Challenge #4: Calculate the New Requirements for Motors

kylesaenz, 2016-03-10:
Go through and calculate what the minimum amount of power is required in order to drive the rover for the required time of 2 hours at 1.5 m/s. Please reference section 3 of the competition rules, and the localization map for more information.

CrisAdam, 2016-03-10:
Will depend on robot weight - assume worst case and 10% incline, and our current 55% efficiency of motor + gearbox

jsspurlo, 2016-04-16:
I made sure to write down the calculations when Nattu taught us. Based on the equations he gave us, I can calculate that, based on the NPC motors and size of wheels, I think we need a minimum 90 amp-hours at 24 volts (2160 watt-hours). For driving ONLY, we need an absolute theoretical minimum of 1800 watt-hours. Of course we are doing more than just driving, and the 1800 watt-hour requirement is only theoretical, it could require more or less, based on a lot of factors. 
 
Assuming the 1800 watt-hour minimum is accurate, having a battery with 90 amp-hours at 24 volts will allow the robot to continuously draw 180 watts to power other devices throughout the entire duration of the 2 hours. This is also assuming the battery is **an ideal battery** that does not diminish in power over time until all the power is used. This is why I first encouraged buying 100 amp-hour batteries, which would give us a theoretical 300 watts of continuous power for other electronics, and would help counter any diminishing of power over time. 
 
If we use any other motor/gearbox/wheel, this would of course change the calculations.