iscumd/UM-Dearborn-Mars-Rover-Challenge #6: Determine grip strenght on collection arm

CrisAdam, 2016-03-10:
In order to not crush the samples, and know when to stop the stepper motors we need some sort of way to measure our squeezing power - suggested item:
[](https://www.sparkfun.com/products/9674)

Also related, add some foam padding or rubber gripping material to aluminum rod

CrisAdam, 2016-03-19:
@olivierurban how much mechanical force would the stepper motors used in the gripping mechanism need to produce (worst case)? (so I can look up their efficiency and scale the electrical side appropriately).

olivierurban, 2016-03-22:
Sorry it took a second, I had to wait for Alex to do the calculation. The motors will require about 0.045 N*m to grasp a 2kg object assuming it is aluminum on steel. When accounting some kind of rubber or foam grippers, the required torque will be about 0.0316 N*m @CrisAdam 
Sorry for the delay

CrisAdam, 2016-03-24:
Thanks - we are good, as the nema 17 motors you have are rated for 23 Ncm, and even if I only power them to 50%, that's still 3x as much as needed.

closed 2016-03-24 by CrisAdam