iscumd/UM-Dearborn-Mars-Rover-Challenge #8: Wheel Research

kylesaenz, 2016-03-10:
Find a set of wheels that can be mounted to the motors, and can be filled with foam.

jsspurlo, 2016-04-16:
I have looked again, and again, and again, and I think we'd have better luck manufacturing our own wheels out of aluminum honestly. 
We should make a CAD model of exactly what we want, 3D-print it in pieces, put it together, make a mold out of it (or just 3D-print the mold pieces), and cast it out of aluminum.
 
[https://www.youtube.com/watch?v=O_TU-PCudoE](https://www.youtube.com/watch?v=O_TU-PCudoE)
 
[image]

CrisAdam, 2016-04-16:
I would also add that in theory there is a plasma cutter in the weld shop that could be used for this purpose